eclipse-opendut/opendut #4: Enablement of edge client (EDGAR) with automtive functionality

Ovomaltine85, 2023-12-20:
One story is about enabling the edge client (EDGAR) with further automotive protocols, i.e. CAN (currently only ETH), and other automotive functionality to communicate/ interact with directly connected DuT(s) or those which are accessible via other EDGARs as part of the VPN network. 

Specific features for this story are:
- [ ] Implement a CAN interface (beside current ETH) on EDGAR to enable CAN communication with the locally connected DuTs
- [ ] Establish a communication link via CAN to other DuTs connected to the same EDGAR. Two or more DuTs connected to same EDGAR should use direct communication (not routed via VPN)
- [ ] Establish a communication link via CAN to other DuTs located at different EDGARs that are part of the IP-routed network (VPN)
- [ ] Implement automotive functionality on EDGAR to achieve following use cases:
        - Trigger automotive clamps (Klemme 15/30)
        - Integrate an (external) restbus simulation
        - Flashing of a DuT via EDGAR
        - General I/O for integration of sensors on DuT

kKdH, 2023-12-20:
Thank ypu @Ovomaltine85 for opening this issue. May you also setup a call so we can discuss the details and refine this issue further?

@Ovomaltine85 @eet1mu3:
Could you add your real name to your GH account? I have no clue who you are. :smile:

Ovomaltine85, 2023-12-20:
> Thank ypu @Ovomaltine85 for opening this issue. May you also setup a call so we can discuss the details and refine this issue further?
> 
> @Ovomaltine85 @eet1mu3: Could you add your real name to your GH account? I have no clue who you are. 😄

Good hint with the clear names ;-) 
Regarding the alignment call. This is definitely a good idea. Do you want to invite or shall we? 
There will be further colleagues working on this story. Not all of them are assigned to this Github org yet.

It may make sense to have the one or other discussion here at GH discussions to make our decision making transparent and documented.

jbaumgartl, 2023-12-22:
Pls invite meetings via our mailing list.