eclipse-opendut/opendut #9: The MTU of wireguard interfaces created by Netbird must be adjustable

kKdH, 2023-12-20:
## MTU - technical background
The default [MTU](https://en.wikipedia.org/wiki/Maximum_transmission_unit) of Wireguard is 1420. In Netbird the default MTU value is 1280 because it accounts for some additional header (140 bytes) that is required by the tunneling feature of netbird. This tunneling feature allows for connecting arbitrary participants/peers over different layers of NAT while making sure that they are communicating with the most direct connection that is possible. Then generic route encapsulation (GRE) requires another 38 bytes of the header to work. The bridged interface with netbird will end up with an MTU of 1242. 

## Network configuration for ECUs
If we want to bridge arbitrary interfaces we need to keep the default MTU at 1500 bytes. Therefore we need to raise the MTU of the wireguard interface by (1500 - 1242) = 258 bytes. 
The final MTU of our wireguard device created by netbird should be 1280 + 258 = 1538 bytes.

How to find necessary MTU to make our setup work? This was evaluated [here](https://git.i.mercedes-benz.com/Distributed-Testing/civic_richos_qnx_scripts/blob/distributed_testing/notes-wireguard.md)

[image]

## Relevant feature request in Netbird
I want to be able to change MTU (similar to vanilla Wireguard or Netmaker), either through local configuration or management UI. See official [issue]( https://github.com/netbirdio/netbird/issues/743).

For us updating the default MTU value might already be enough:
https://github.com/netbirdio/netbird/blob/6c2b364966c7dd0f6eaed0cd3f8fa378d757f319/iface/iface.go#L16
```
const (
	DefaultMTU    = 1280 // should be 
	DefaultWgPort = 51820
)
```

## pull-request and pre-release
@reimarstier has already opened a pull-request: https://github.com/netbirdio/netbird/pull/1101

Until the changes are available upstream a custom release can be found here:
https://github.com/reimarstier/netbird/blob/0.22.7-357aab4/client/proto/daemon.proto
https://github.com/reimarstier/netbird/releases/tag/0.22.7-357aab4

ThHirsch, 2024-03-25:
Any chance, in getting thi into the newer version.
I was not able to get Netbird to properly handle https to an Endpoint, if my client is/was conntected to a network which connects to the internet by a DOCIS3-1 cable modem.
It seems the packets either get fragmented (which is rendering the https unresponsive) or dropped then. In any case I can ping, but won't get any (secure) connection that is going to send 'bigger' content.
Fiddling with the MTU in router or the interface by itself doesn't seem to help either, so it seems to be the netbird client by itsself which needst to be 'tuned' for the MTU.

kKdH, 2024-03-25:
Hey @ThHirsch, we also looking forward to get this feature soon, but we cannot force the netbird developers. We are just netbird users too.

Within openDuT we use a custom build of netbird containing changes from @reimarstier to adjust the MTU to our needs.

If you want, you can comment [here](https://github.com/netbirdio/netbird/issues/743) to let the netbird developers know, that there are more users waiting on that feature and maybe they will raise the priority.

reimarstier, 2025-07-02:
Hey @ThHirsch, [Eclipse OpenDuT](https://opendut.eclipse.dev/) and [Netbird](https://netbird.io) are two different projects. There is no ongoing work on this subject from our side since we don't have the resources to get more involved in the Netbird project. From what I gather from the linked issue there are some interested parties for this feature but no one is actually working on a contribution.

My pull request to the NetBird project was deemed [incomplete](https://github.com/netbirdio/netbird/pull/1101#issuecomment-1787301098). The main issue the Netbird developer raised was the fact, that the respective buffers need to be adjusted and the fact that it only makes sense to adjust the MTU if all members of the group have the same MTU. In OpenDuT we ensure the MTU is set by our client but it would be definitely better if the MTU was set via a group configuration in NetBird.