eclipse-opendut/opendut #13: Test Manager as central test application in openDuT

Ovomaltine85, 2023-12-21:
This story is about the Test-Manager (TM) component that represents the central application in openDuT to create new test projects, connect DuTs with specific test cases, schedule/ execute test runs and generate test reports.

Following features are to be integrated in a first step:
- [ ] Import and parsing of test cases provided in different formats (DSL, yaml, Robo Framework etc.) - consolidate solutions between ETAS and AVL 
- [ ] TM receives a representation of the DuTs provided by EDGAR and CARL (e.g., available interfaces, other I/O, power supply) and thereby makes DuTs usable in TM
- [ ] TM communicates with EDGARs via VPN provided by CARL for
      - Receiving test results that were executed locally (cf. #5)
      - Provision of test engines (containers) to EDGAR for local execution
      - Provision of test setups/configuration to EDGAR for preparing local test setup (rest bus simulation, power control, clamps etc.)
- [ ] Execution, scheduling/planning and control of new test runs based on parsed test case (description) - consolidate solutions between ETAS and AVL
      - Case 1 (prio): Tests are executed locally on EDGAR - consolidation of test results and reporting by TM
      - Case 2: Tests are executed centrally by TM and test messages/ data is relayed to EDGAR, TM generates report
      - Case 3: Integration of external/remote test engines with openDuT (SaaS)
- [ ] Implement a user management on TM with RBAC (test engineers, project managers, auditors etc.)

kKdH, 2023-12-21:
> Implement a user management on TM with RBAC (test engineers, project managers, auditors etc.)
this should be aligned with #16

hafklin, 2024-05-29:
@Ovomaltine85 as I understood, the ideas described here are no longer wanted, correct? In that case, shall we close this issue?

Ovomaltine85, 2024-05-31:
You're right @hafklin  
Test Manager is at the moment not in focus of implementation.
However, we have to re-evaluate in the future how we manage external test triggers, scheduling of testing and user interaction.

closed 2024-05-31 by Ovomaltine85