eclipse-opendut/opendut #24: IPv6 support

eet1mu3, 2024-01-10:
https://github.com/eclipse-opendut/opendut/blob/90a4bfeb7e382ece7c56d11b151f50e002423a2f/opendut-edgar/src/service/network_device/manager.rs#L80

reimarstier, 2024-01-10:
Hi, thanks for reaching out and reading the code. The IPv4 supports is only relevant for the virtual private network provided by netbird. We are still able to bridge any kind of layer two (or above) traffic over this private network including IPv6. Therefore, it is not a restriction to our use case. Unless netbird is changing the address assignment we don't need to change this at all.

eet1mu3, 2024-01-10:
Hi. Yes. I agree this should be low prio for now. Seems like netbird is slowly moving in that direction, so we should at least keep that in mind:
https://github.com/netbirdio/netbird/pull/578