eclipse-opendut/opendut #33: CARL should validate bearer tokens contained in requests

kKdH, 2024-01-18:
CARL should validate bearer tokens contained in requests against Keycloak to protect its routs.

mirenz1, 2024-04-10:
To be checked:
  - access token signature
  - token validity
  - token valid for user

token header contains certificate id used for signature validation
caching in memory

reimarstier, 2024-08-01:
This branch was rebased locally, merged locally and development was pushed before the feature branch. Therefore the commit id did not match when code went into the development branch. Consequently this pull request is wrongfully flagged as being rejected/closed despite the fact that the code was merged.

Note to future self:
* First rebase _AND_ push feature branch
* Then merge to development