eclipse-opendut/opendut #53: Complement THEO with an MQTT broker

kKdH, 2024-01-30:
We plan to use MQTT for data exchange purposes between DuTs and peers on the one hand and CARL and other services on the other hand. Examples are:

 - Current and voltage measuring and analyses
 - Control of Actors.
 - Read/Write signals

Because of this, THEO should be extended to provide an MQTT broker integrated into the current setup.

- [ ] Select an appropriate MQTT Broker which matches our requirements.
      - e.g. Keycloak integration.
- [ ] Integrate the selected Broker into THEO

kKdH, 2024-02-01:
Before we start to integrate MQTT into openDuT we should consider kuksa.val as data and signal hub. #59

eet1mu3, 2024-02-20:
[rumqtt](https://github.com/bytebeamio/rumqtt/tree/main/rumqttd) could be a possible contender. I have it working as a demo in standalone mode, but could be also integrated.