eclipse-opendut/opendut #59: Evaluate kuksa.val as hub for data and signal exchange

kKdH, 2024-02-01:
tbd.

tspreck, 2024-02-05:
Hej @erikbosch & @SebastianSchildt take a look and maybe we can suggest some idea's for implementation's

kKdH, 2024-02-05:
This issue doesn't provide much context. If you want, we can setup a call.

SebastianSchildt, 2024-02-05:
Coming across this project for the first time. Speaking of HIL/SIL, for tests of "VSS-level" or similar interfaces this certainly seems a good idea. We also have some mock and replay support that might come in handy here. @eriksven ist likely the right guy for giving an overview.

When this project (again, only read the headline :D) , also thinks about lets say replaying raw Bus traces (e.g. CAN/LIN), that are not processed via KUKSA/VSS (as much as this saddens us as KUKSA developers :D), it might not be a good idea to shoehorn them through some "fake" VSS datapoints, but rather using some lower level way such as virtual interfaces might be better. 

But then again, there are different levels of testing, so maybe even both is needed.

eet1mu3, 2024-02-20:
Virtual interfaces are covered with different technologies, for CAN, something based on Cannelloni: https://github.com/eclipse-opendut/opendut/pull/85

closed 2024-03-07 by eet1mu3