eclipse-opendut/opendut #68: CARL's client to communicate to the Netbird Management API should provide more configuration options

kKdH, 2024-02-06:
The client used by CARL to communicate to the Netbird Management API should provide more configuration parameters:
- The default request timeout.
- The CA certificate to use when connecting to the API
  - This will remove the requirement for `https_only=false`, so the corresponding configs in `carl.toml`, THEO etc. can be removed.
- The expiry duration of a generated setup-key.

The exiting config files should be adjusted to reflect these changes and should provide sensible default values.

kKdH, 2024-02-07:
Is is always the same CA certificate? To be more precise the same CA certificate used for CARL and Netbird? If so, then we should load it in CARL and pass it down to the vpn client, because we need the same cert for #70

kKdH, 2024-02-07:
Updated issue #70 to allow different CA certificates.

reimarstier, 2024-02-22:
Open tasks are addressed in new issue #95.

closed 2024-02-22 by reimarstier