eclipse-opendut/opendut #105: CAN integration

reimarstier, 2024-02-27:
As soon as the CAN support (#85) has landed, there is further work to be done:

* Make timeouts configurable where necessary (real setups may have latency issues when network delay between peers comes into play, buffer_timeout 1 microsecond, response timeout 2 seconds, currently only tested with virtual can within the same machine almost no delay measurable)
* Add configuration of CAN devices via LEA
* CAN health check (ping?), round trip time
* Address assumption in code: Any interface containing can* is a CAN interface
* Make CAN/ETH optional (e.g. peer with only CAN interface or peer with only car Ethernet interface)

@mbfm @hafklin Feel free to modify/add what you have on your mind

hafklin, 2024-02-27:
Thanks for collecting the TODOs here. Two more things that need to be done:

- Currently, the port on which cannelloni listens on the leading EDGAR is derived from the client IP address. This may in some cases lead to port collisions and should be handled properly by having CARL assign the ports to the EDGARs.
- Currently, it is assumed that all physical CAN devices to be used with an EDGAR are already configured (bitrate etc.) and up. However, it may be desirable to allow the user to specify these configuration parameters via LEA/CLEO and have EDGAR configure the CAN devices itself.

> CAN health check (ping?), round trip time

@reimarstier where would you like to see this? A basic version of this already exists [here](https://github.com/eclipse-opendut/opendut/blob/development/.ci/docker/edgar/scripts/pingall_can.py) and is executed within the test environment.

reimarstier, 2024-02-28:
Regarding the health check, I was thinking about EDGAR performing this action regularly. This does not exist yet and is not restricted to CAN.

mirenz1, 2024-08-28:
See also #164