eclipse-opendut/opendut #115: Implement automotive functionality on EDGAR to achieve following use cases:

mrh1234, 2024-03-01:


mrh1234, 2024-03-01:
General idea:

Restbus simulation:
- Implement ARXML parser to filter out CAN messages relevant for restbus simulation
- Execute restbus simulation by utilizing Linux Kernel CAN Broadcast Manager

Flashing:
- TBD

mrh1234, 2024-03-06:
> This library might be very useful: https://crates.io/crates/autosar-data

I will stick with programming my own parser. The time-intensive operations of my parser are already implemented and take less than 1 minute for parsing a 325 MB example ARXML file. With optimizations at a later stage, I might even decrease that time. In comparison, the autosar-data library takes more than 6 (!!) minutes for parsing the same file.

kKdH, 2024-03-07:
Hey @olFi95 did you face similar performance problems when parsing ARXMLs with [autosar-data](https://crates.io/crates/autosar-data)? Do you have some tips?

mrh1234, 2024-03-18:
I had a bug in my time measurements and the autosar-data library is indeed performant. So, I will use it.

mrh1234, 2024-05-02:
Status update: ARXML parsing itself is implemented in the restbus-simulation branch. Next, I have to decapsulate the parser a little bit and start to integrate into into Edgar. Currently, it is independent of everything. Furthermore, I have to setup the restbus simulation that takes the output from the parser and plays the messages for a given CAN cluster/bus. This is currently under work.