eclipse-opendut/opendut #139: openDuT MVP - user showcase

Ovomaltine85, 2024-03-14:
This issue is about describing and tracking the steps for creating a minimal viable product (MVP) to showcase the value, functionality, and integration of openDuT for a concrete automotive test case.

## Overall goal
To convince users and potential customers of openDuT, we want to create a minimal viable product (MVP) which shows the general functionality of the framework by executing a simplified test scenario.

### User story:
As a (potential) user of openDuT
I want to verify if openDut (basically) can satisfy my needs of an automotive test framework
So that I can test the integration of openDuT into my DevOps, run simple test scenarios and store the test result at a defined place.

### Minimum requirements (must have)
- [ ] openDuT interacts via CAN and ETH with DuTs that are connect to one single peer.
- [ ] DuTs that are connected to one peer shall be able to communicate with each other via CAN/ETH
- [ ] The test case (TC) describes in an agnostic language (DSL) the test steps by individual instructions
- [ ] Test scenario: Execute a port scan (ETH) of one DuT by nmap inside a Docker container. The docker image is available locally on the peer.
- [ ] The test execution instruction (TEI) shall be compiled out of the TC 
- [ ] The peer shall perform (test executer) the test scenario based on the TEI
- [ ] The test result (log, string etc.) shall be stored in a repository, e.g. openDuT Github
- [ ] openDuT shall provide continuously status information about the DuT, like current/ voltage, bus load and shall be able to trigger power-cycles or Klemme XY. The status information shall be accessible via a dashboard (e.g. refreshing html page).

### Additional requirements (may have)
- [ ] DuTs that are connected to one peer may also interact with other DuTs connected to other peers via CAN and ETH
- [ ] Additional test scenario: Execute a python script locally in virtualenv on peer that performs a simple UDS scan
- [ ] The framework may provide a (rudimentary) rest bus simulation to keep DuTs in a testable state
- [ ] openDuT may provide a (simple) GUI that supports the user with creation and edition of a TC
- [ ] The docker may be pulled from an external container registry 

## Time frame
The completion of the MVP (minimum) is projected for mid/end of April'24.

## ToDo's
- Link tasks above with already existing issues/ stories here in Github
- Link PRs to implement the MVP with this issue to keep track of progress

hafklin, 2024-03-25:
PRs and issues related to the implementation of CAN functionality:
#85 - Add CAN Support
#105 - CAN Integration
#128 - Allow configuring CAN interfaces via LEA/CLEO
#129 - EDGAR -> Use Cannelloni ports provided by CARL
#136 - Add configuration to network interfaces

Ovomaltine85, 2024-04-02:
Linking with #54 , where software execution environment is being developed for EDGAR (podman + docker).
@LenaJok , @sed7be for reference