eclipse-opendut/opendut #171: CAN bus openDuT network example setup

mirenz1, 2024-04-02:
Validate CAN-Test example setup:
peer1 (test script on CAN) --> openDuT tunnel --> peer2 (CAN interface)

- [ ] CARL setup (VM, "DEV-box")
- [ ] peer configurations via CLEO scripts
- [ ] roll out EDGAR on peers
- [ ] test script for sending/receiving CAN message

reimarstier, 2024-04-18:
At the moment there are a lot of configuration parameters to be set that make automation rather tedious. The test environment does a fine job of hiding this complexity. 

To use this CLEO in different setups and remove a lot of the manual configuration tasks to have a CLEO instance connect to a new CARL instance, a few other issues need to be done first.

Therefore this issue relates to the following issues:
- #190 
- #193

reimarstier, 2024-04-25:
There are currently a few known issues with the current setup that are being addressed (WIP) in branch issue-171-revised:

- #186
- [ ] Create PKI in container with predictable user id and group id
  currently owned by the user who cloned the repository, carl.key only readable by user id of host system

voelkera, 2024-05-08:
Validate CAN-Test example setup:
peer1 (test script on CAN) --> openDuT tunnel --> peer2 (CAN interface)

- [ ] CARL setup (VM, "DEV-box"), relates to https://github.com/eclipse-opendut/opendut/issues/151
- [ ]  setup CLEO for CARL (provide as container in registry)
- [ ] provide EDGAR release for x86 and ARM64 target
- [ ]  roll out EDGAR on peers
- [ ] provide Cannelloni build for for x86 and ARM64 target
- [ ] test script for sending/receiving CAN message

mtwardawski, 2024-05-08:
- [x] Create PKI in container with predictable user id and group id
currently owned by the user who cloned the repository, carl.key only readable by user id of host system

mtwardawski, 2024-05-10:
[x] if there is already an existing ca.pem and netbird certificate, new certificates from setup string cannot be used in EDGAR managed setup. https://github.com/eclipse-opendut/opendut/issues/224

mtwardawski, 2024-05-10:
- [x] In case of development mode, EDGAR performs a dry-run which looks like actually setting up the peer.
This is misleading. The note that EDGAR is running in "Dry-run mode" should be more prominent (banner, color, ...).

mtwardawski, 2024-05-15:
- [x] Architecture Overview in documentation

mtwardawski, 2024-05-21:
We achieved to get a connection between two EDGARs now:
- [x] roll out EDGAR on peers

PR #225 (2024-05-27): New Deployment Infrastructure (Issue 171)
Closes #171.
closed 2024-06-27 by reimarstier