eclipse-opendut/opendut #186: Build cannelloni for same target as EDGAR and provide option for rollout together with EDGAR

mirenz1, 2024-04-10:
When EDGAR is rolled out on a peer it also needs to have cannelloni installed on the peer in order to be able to provide CAN bus connectivity.

- [ ] Action workflow to build cannelloni
- [ ] Option to roll out EGDAR together with cannelloni, similar to the Netbird client
- [ ] start/stop additional services (cannelloni, netbird client, ...) together with EDGAR

hafklin, 2024-04-11:
Hi Michael,
what you propose would definitely be convenient, I just wanted to note that I believe we need to be a bit careful with the GPLv2 license under which cannelloni is published (which seems to be incompatible with the Apache license of OpenDuT). I'm not sure whether the ideas above would in some way violate GPLv2 but it should definitely be checked before.

mirenz1, 2024-04-11:
Hi Linus,
thanks for your comment.

From what I understand, we should check
- that we can provide the cannelloni source code from where we build it
- that we can declare any cannelloni source code modifications
- how we integrate openDuT with cannelloni, e.g. 
  - integrate cannelloni code in EDGAR binary (built-in) --> not the best option, and as far as I understand not needed
  - integrate cannelloni as statically linked software --> should be avoided
  - integrate cannelloni as dynamically linked software
  - no integration, EDGAR calls cannelloni executable

reimarstier, 2024-04-29:
Releases of cannelloni binary will be published here: https://github.com/eclipse-opendut/cannelloni/releases/