eclipse-opendut/opendut #193: Download CLEO endpoint

reimarstier, 2024-04-12:
To run CLEO in CI/CD a user wants to always have the matching binary for the current architecture.

### Tasks
- [ ] CARL should provide an HTTP endpoint to download CLEO
  * Serve debug binary in development mode
  * Serve directory with release binary of CLEO for the requested architecture
- [ ] A distribution build should include the CLEO release within CARL's distribution
- [ ] Add documentation on how to configure your CI pipeline

### Details
A user wants to use CLEO in CI/CD. The following could be a description on how to use it.

Configure workflow environment variables:
- CARL url
- CARL CA (Keycloak CA, if different)
- issuer URL for keycloak
- Client id, secret
```
echo $CARL_CA > /path/to/ca.pem
export SSL_CERT_FILE=/path/to/ca.pem
curl https://carl/download/arch-triple/opendut-cleo.tar
tar xf opendut-cleo.tar
./opendut-cleo
```

mtwardawski, 2024-04-19:
- [ ] CARL should provide an HTTP endpoint to download CLEO
  - [ ] Serve debug binary in development mode
  - [ ] Serve directory with release binary of CLEO for the requested architecture
 - [ ] A distribution build should include the CLEO release within CARL's distribution
 - [ ] Add documentation on how to configure your CI pipeline

mirenz1, 2024-05-02:
Currently the download does not provide the client certificate, clientID and client secret.
Requires: #190 and #198 to be fully usable