eclipse-opendut/opendut #242: Cannelloni should be built with SCTP support

reimarstier, 2024-05-28:
- [x] Add SCTP support to cannelloni by including library in bundle
- [ ] Update documentation:
  - [ ] Installing SCTP is no longer required
  - [ ] Check if licensing requirements are met (e.g. Sources.md)
- [x] Check newly created binaries run on the following platforms (armv6, arm64, x64)
- [ ] Check if newly created binaries work for our setup

The architecture armv7 is currently untested due to lack of hardware with soft-float installation (raspbian is hardfloat).

mirenz1, 2024-06-26:
Add cannelloni source code to cannelloni package

reimarstier, 2024-06-27:
Added source code to cannelloni package. Tested the following packages on the hardware:
* cannelloni_linux-arm64-lts_1.1.0.tar.gz
* cannelloni_linux-armv6-lts_1.1.0.tar.gz
* cannelloni_manylinux_2_28-x64_1.1.0.tar.gz

closed 2024-06-27 by reimarstier