eclipse-opendut/opendut #249: Replace the usage of update-ca-certificates by another solution

mtwardawski, 2024-06-11:
Issue is in WIP - will be enhanced with more information.

Currently in task WriteCaCertificates we are using the command `update-ca-certificate`. This command will not work on some Linux distributions, so EDGAR cannot be started on those distributions and tests will fail in `cargo ci check`.  This happens around line 119 ff in `write_ca_certificates.rs`.

```
let update_ca_certificates = which::which("update-ca-certificates")
        .context(String::from("No command `update-ca-certificates` found. Ensure your system provides this command."))?;

command_runner.run(
    &mut Command::new(update_ca_certificates) //Update OS certificate store, as NetBird reads from there
).context("update-ca-certificates could not be executed successfully!")?;
```

TODO: Provide possible solutions later

mirenz1, 2024-08-14:
Relates to [#263](https://github.com/eclipse-opendut/opendut/issues/263)

mohammadp1001, 2026-03-22:
Hello, I would like to work on this issue if it is still relevant.