eclipse-opendut/opendut #306: Evaluate open1722 as replacement for cannelloni

mirenz1, 2024-10-02:


SebastianSchildt, 2025-02-13:
I think  - as a first step for evaluation - replacing cannelloni with an Open1722 example executable seems quite doable

This is a bit theoretical, I do know Open1722 a bit bit but I currently have no OpenDUT setup and never compiled this stuff, so take with a grain of salt.

Basically I think the code in question for starting cannellon instances is here

https://github.com/eclipse-opendut/opendut/blob/c41a3334e59a31f28feb8473a0c5bf7345d7ec9d/opendut-edgar/src/service/cannelloni_manager.rs#L131-L149

so for a “server” if I parse it correctly it might do

```
cannelloni -I <SOMECAN> -S s -t <SOMETIME> -R <REMOTE_IP> -l <SOMEPORT> 
```

for client it does

```
cannelloni -I <SOMECAN>  -S c -t <SOMETIME> -R <REMOTE_IP> -r <SOMEPORT> 
```

Now in 1722 - also running on IP I think the example acf-can-bridge can achieve the same

One EDGAR

```
acf-can-bridge --canif <SOMECANA> -u -p <SOMEPORTA> --dst-nw-addr <REMOTE_IPB:SOMEPORTB> 
```

other EDGAR

```
acf-can-bridge —canif <SOMECANA> -u -p <SOMEPORTB> --dst-nw-addr <REMOTE_IPA:SOMEPORTA> 
```

Also pinging @nayakned

I would hope this is "as good as cannelloni"; if not Open1722 is interested to know.

If it works, there are many options to "optimize" it, e.g. the functionality could be build/linked in  directly by using Open1722 API functions. Also as this is UDP based - or if desired even its own ethertype without IP - MAYBE by using multicast groups or MAC addresses some efficiency might be gained in some situations, but I think that is more of a general networking nut to crack (how would the GRE/Wireshark setup play with this? At what points would/should the packets be duplicated for best efficiency/performance)

Just dropping support via the exmaple bridge would already give
 - a standardized way to encapsulate CAN in Ethernet (whether that has any "worth" is of course in the eye of the beholder)
 - As it as expected to see this technology also in "real ECUs", it makes it easier in a mixed HIL/SIL setup to integrate real ECUs in the OpenDUT setup

Also pinging @nayakned

SebastianSchildt, 2025-02-13:
Would also solve/make #251 obsolete, as Open1722 is BSD licensed

hafklin, 2025-04-02:
I claimed this issue for now, as someone from our side will be working on this soon.