eclipse-opendut/opendut #346: CLEO network interaction

DocBO, 2025-04-29:
New Feature proposition, CLEO as message broker:

- we want to send and broadcast Messages through the CLEO interface into a selected deployed network allowing to trigger communication from the ECU testing point of view.
- we also want to receive ongoing broadcast network communication through CLEO interface.

kKdH, 2025-04-29:
We could assist you better if you shared more details about your use case.

DocBO, 2025-04-29:
Thanks for the reply. I will evaluate this approach. Linux is of no problem and change of network is supposed to be of no problem. 
Use case would be: we want to use communication to network devices to build up a testing strategy. And the communication layer is kind of just an intermediate layer. I have seen that you implemented peer-messaging-broker.proto (belongs to carl i guess). But to use it we need all the network protocol setup which is already done inside the CLEO interface. So why not to use a message broker on top of that? That was just an idea. If there is a standard or better approach to do this, would be fine for me too.

kKdH, 2025-05-08:
You are right, there is the `PeerMessagingBroker` gRPC services within CARL. This services is used as a communication hub for internal communication between CARL and the connected EDGARS. Currently, this communication is limited to sending `PeerConfiguration` from CARL to EDGAR and status information from EDGAR to CARL. This dedicated communication channel is not designed for transmitting arbitrary messages, and the `PeerMessagingBroker` should not be misinterpreted as an MQTT broker.

kKdH, 2025-05-15:
I'm going to close this issue. Please don't hesitate to reopen it or create a new one if needed.